Strooom/GRBL-Post-Processor #3: Provide a safe fallback to WCS1 / G54 when no WCS is set in CAM

Strooom, 2016-09-05:
In order to use the correct set of World Coordinate System, you should set it in your CAM-process.
The PostProcesser then translates this setting (WCS1..WCS6) to G54..G59
When not set, it will raise a warning.
However, as many (third party) libraries are not setting a WCS, the PostProcessor should **default** to WCS1 / G54 in that case..

Strooom, 2016-09-05:
Solved

closed 2016-09-05 by Strooom

Bigfellas, 2017-01-06:
I am having trouble setting WCS in fusion 360. I get an error asking me to set WCS1..6 even if I have setup a corord system in fusion. Is there a particular way to set it. Thanks
Matt

Strooom, 2017-01-06:
Hi, 

in the CAM environment, right-click on the Setup, then choose the 3rd tab 'Post Process'
Here you select WCS Offset, typically nmbr 1 = G54.
I don't know why Fusion360 doesn't put a 1 here as default, as 0 doesn't make sense on the CNC..

Have a nice day!

[image]

Bigfellas, 2017-01-06:
ahh thanks so much . i thought that was a unit of offset. like 1mm. thanks for clearing that up for me.
i am having other problem with your post process. at the start the z axis is moving down. if i use the fusion grbl process it works.. could this just be the problem with the wcs not being setup properly..

thanks Matt

Strooom, 2017-01-07:
Hi Matt, 
I added another Wiki page 'Anatomy of output of the Post Processor'. 
Please check this page, it should answer your questions. If not, feel free to come back for more help!

Bigfellas, 2017-01-07:
Amazingly helpful. Thanks So much. I have forked this and made a few tweaks now for my machine. 
Really appreciate you doing this. Thanks.