Strooom/GRBL-Post-Processor #5: Laser Mode

kikolobo, 2016-12-05:
Hi,

Would it be possible to create an option to set for Laser mode, where you set the number of passes and the Z axis deltas (lowering) per pass. And also set the laser power instead of the spindle speed. 

Some GRBL firmware use spindle speeds of say around 0 - 12000. And that translates to laser power (PWM).  Others hace 0 - 255. So a max laser power option should also be available.

See https://jtechphotonics.com lasers.

A good ref is their inkscape plugin:
https://jtechphotonics.com/?page_id=2012

Strooom, 2016-12-06:
Hi @kikolobo 
I can take a look at it, but I don't understand your request 100%.

The spindle speeds are set in Fusion in the CAM operations. What I could do is:
* have a MaxSpindleSpeed setting (you can set it eg to 10000)
* then have a LaserMode setting which translates spindle speeds to a range of 0 - 255

As for your request about several passes and lowering Z for each pass: this is also done in Fusion360. You (eg.) create a 'Contour' operation, then in the 'Passes' tab of that operation, activate 'Multiple Depths'

The purpose of the Post-Processor is to generate machine compatible GCode from the CAM operations. It should (imho) not convert a single pass into multiple passes, as this is the task of the CAM and Fusion360 already does this very well.

Anyway, let me know what I can do, and I am certainly willing to help you.

kr,

P

kikolobo, 2016-12-06:
You are correct!

Let me give it a try and get back to you with more information on what would work for laser mode!

Thanks!

closed 2017-02-12 by Strooom