Strooom/GRBL-Post-Processor #10: Fusion 360 post process order

vMeph, 2017-07-06:
@Strooom  in fusion 360 post processor is there anyway to make order of actions automatic

for exemple on start usualy we put the  bit toutching the surface of the job we doing has a start point, and then start the job,  and the order fusion posts is first turn the drill, and then got to a position and only then raises Z
this causes the drill to be on and by moving to position and only then raise the Z, will scratch surface of material

G90 G94
G17
G21

(Operation 1 of 4 : 2D Contour1)

G53 G0 Z0
G54
Z10                          <---------------change this become the S10000 M3 
G4 P0.8
X32.918 Y14.38
S10000 M3              <----------------change and this the Z 10
Z5.015
G1 Z2.015 F150
Z-0.775

at the end of g code same

G0 X-1.68 Z10

G53 Z0                     <-----------------change this become the M5
M5                            <---------------- change and this the Z10
G4 P0.8
G53 X0 Y0
M30

is there anyway to achive this?

vMeph, 2017-07-09:
i was able to change the order, ill leave here not sure if will help
can you review
[OpenbuildsGRBLv1.txt](https://github.com/Strooom/GRBL-Post-Processor/files/1133874/OpenbuildsGRBLv1.txt)
thanks

Strooom, 2017-07-09:
Hi,

Yes, it should be simple to change the order, but I think it is not
necessary.
I think you are maybe not using Fusion CAM heights in the intended way.
I will investigate and explain later this week when I have more time.

Tnx for your msg and for supporting the GRBL Post Processor.

Pascal

vMeph, 2017-07-09:
@Strooom thanks for answer
in fusion i set height
for exemple
clear height off set to 10
retract height offset to 5
feed height off set to 5
top height to 0 from stock top
bootm height 0 from stock bottom

but for some reason always post the same order
(Operation 1 of 4 : 2D Contour1)

G53 G0 Z0
G54
S10000 M3 
G4 P0.8
X32.918 Y14.38
Z10 
Z5.015
G1 Z2.015 F150
Z-0.775

so the drill always turns on first and only then moves
i was able to make order changes on text above i sent, but still would like to know why the post first turns drills on and then does the moves to positions
if you clould clarify me on that would rox, maybe doing something wrong

thanks

Strooom, 2017-07-10:
About the selection of Work Coordinates : Fusion takes this from the selection you make on the CAM Setup : 3rd tab : Post Settings. Default (in my experience) this is set to 0, but GCode actually expects a value from 1 to 6.

So when you create a **setup** : simply set this WCS value to 1 (on the third setup properties tab) and the Generated GCode will understand that you are using G54 WCS.. It is possible to work with templates in Fusion, which could then be set to WCS, so every new design, you already have WCS 1, instead of the invalid WCS 0.

Strooom, 2017-07-10:
Hi vMeph, have you checked the Wiki-page : https://github.com/Strooom/GRBL-Post-Processor/wiki/Anatomy-of-output-of-the-Post-Processor ?

This explains the sequence and it's logic.
When you say the drill scratches the surface, something is wrong because normally the generated GCode moves the spindle UP all the way, before moving to the starting point, then activates the spindle, then starts going down.

So I see 2 possible causes : 
1. your parameters for the home position are incorrect, and the home Z is too low,,,
2. your controller (other than GRBL ?) does not execute the G53 (correctly) - I have sees this on some machines. In this case the Spindle does not go up first, and indeed may cause a problem (crash) when moving to the start of the first operation..

I hope with this info you can solve your problem. But as he posted GCode does things in the correct sequence, I don't see it needed to change the Post-Processor for the time being.

closed 2017-07-10 by Strooom

vMeph, 2017-07-10:
@Strooom
Question
Wiki link says
From here on the output is the toolpath as generated from Fusion360. Fusion360 first moves in XY, then moves down, usually in several steps : In this example the job starts at X = -142.25, Y = -142.25 then it moves down to Z = 80, (to what is set as clearance or retract height in Fusion360) then Z = 60, (to what is set as Feed height in Fusion360) finally it moves down to Z = 54 at feedrate speed 500 mm/min (now it is at the Z to start the toolpath)

X-142.25 Y-142.25
Z80
Z60
G1 Z54 F500

Confused now about this part 
job starts at X = -142.25, Y = -142.25 then it moves down to Z = 80, (to what is set as clearance or retract height in Fusion360)

Moves down to Z= 80 positive? Should it be z = -80, negative

vMeph, 2017-07-10:
Maybe the way im doing things is wring
I put my drill bit on where i want to start my job, drill bit toutching the material and position is x0 y0 z0, i start job drill turns ond and the machine moves to the x y and then moves to z height up comes down and mills,

Im very confused in how im doing thing
My z going up is positive and coming down negative

Strooom, 2017-07-11:
Z up is positive, Z down is negative - that is common practice.
Further the Work Coordinates X=0, Y=0, Z=0 are where you position your Setup in Fusion CAM. This does not need to be the starting point of the first operation..

Search Youtube from some good getting started tutorials on Fusion360 CAM. Search for 'Lars Christensen'
I think you may be missing a few essential pieces of the puzzle.

vMeph, 2017-07-12:
@Strooom now i fell silly i found that the problem is that i would go to the place i wanted to mill and the i would set the 54 position but then i would start job with out moving the Z up 
so what would happen is

1- start spindle                 (would scratch material)
2 -move X Y  to position   (would scratch material)
3  - move Z up
4 - move Z down and start mill

eaven knowing it might sound silly what i was doing the logic of order in how do things dont seems to have logic to me
the logic should be

1 - move Z up            
2 - go to X Y position 
3 - turns the spindle On
4- move Z down and start mill