Strooom/GRBL-Post-Processor #12: G53 G0 Z-10  down in material

maumontel, 2017-11-21:
The first line of the code that real move the machine G53 G0 Z-10 send the machine 10 milliliters down to the material,  then move to the initial point, then up 15 milliliters down 10 and now the work is begins as wished.

Could someone give me some light, where could I change this behavior? 

An it does the same at the end G53 Z-10

**The initial part of the code**

G90 G94
G17
G21

(Operation 1 of 1 : Face3)

G53 G0 Z-10
G54
S10000 M3
G4 P0.8
X43.75 Y1.283
Z15
Z5
G1 Z0.4 F1000
G18 G3 X43.45 Y1.2829 Z0.1 I-0.3
G1 X41.8 Z0.1

**End of the code**
G0 X-0.3 Z15

G53 Z-10
M5
G4 P0.8
G53 X-10 Y-10
M30
%

Strooom, 2017-11-23:
Before moving to the starting position of the first operation, the spindle is raised all the way up (in fact to Machine Home Z position). This is done in function onSection(), line 306:
```
	if(isFirstSection())
		{
		writeBlock(gAbsIncModal.format(90));	// Set to absolute coordinates
		if (isMilling())
			{
			writeBlock(gFormat.format(53), gMotionModal.format(0), "Z" + xyzFormat.format(properties.machineHomeZ));	// Retract spindle to Machine Z Home
			}
		}
```
At the end of the job, the same is done to move the spindle away from the workpiece, in a safe way. This time the code is found in onClose(), lines 505 (moving spinde Up) and 513 (moving the spindle to home X-Y)

With this info, you should be able to modify this behaviour.

```
function onClose()
	{
	writeBlock(gAbsIncModal.format(90));	// Set to absolute coordinates for the following moves
	if (isMilling())
		{
		writeBlock(gAbsIncModal.format(90), gFormat.format(53), gMotionModal.format(0), "Z" + xyzFormat.format(properties.machineHomeZ));	// Retract spindle to Machine Z Home
		}
	writeBlock(mFormat.format(5));																					// Stop Spindle
	if (properties.hasCoolant)
		{
		writeBlock(mFormat.format(9));																				// Stop Coolant
		}
	onDwell(properties.spindleOnOffDelay);																			// Wait for spindle to stop
	writeBlock(gAbsIncModal.format(90), gFormat.format(53), gMotionModal.format(0), "X" + xyzFormat.format(properties.machineHomeX), "Y" + xyzFormat.format(properties.machineHomeY));	// Return to home position

	writeBlock(mFormat.format(30));																					// Program End
	writeln("%");																									// EndOfFile marker
	}
```

closed 2017-11-23 by Strooom

maumontel, 2017-11-26:
Hi Pascal,

I find where the Z-10 came from.
It was **machineHomeZ** settings in post-processing screen, everthing was -10 before start

Thank you for your attention,
Mauricio
 
[image]