Strooom/GRBL-Post-Processor #13: Units choice

ENGR-mike, 2017-12-10:
I was under the impression you fixed the post processor to use either mm or inches.  It does not allow the option.

Strooom, 2017-12-10:
As far as I understand it should work as follows : 
* let fusion output inches, this should insert a G20 in your gcode, io G21
* set the post-processor option to inches as well, so it knows you are intentionally sending G20, and not only by mistake..
* machineHome will also be interpreted as inches

GRBL will translate your numbers to the correct number of steps

closed 2017-12-14 by Strooom

ENGR-mike, 2017-12-15:
Sorry, but this issue is not closed.  Today was the first time I was able to check the process.
Starting with a model file in inches and a CAM setup in inches, I selected inches for units in the post processor dialog.  
[image]
Unfortunately, I still received the following error:
[image]

vMeph, 2017-12-15:
Do you have your grbl set on inch mode on configs $$?

ENGR-mike, 2017-12-16:
Please be more specific.  What do you mean by “grbl set on inch mode on configs $$”?

From: Meph1
Sent: Friday, December 15, 2017 2:46 PM
To: Strooom/GRBL-Post-Processor
Cc: ENGR-mike; Author
Subject: Re: [Strooom/GRBL-Post-Processor] Units choice (#13)

Do you have your grbl set on inch mode on configs $$?

vMeph, 2017-12-16:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration
$13=0 metric mode
$13=1 inch mode

Strooom, 2017-12-16:
In fact there was some misunderstanding from my side about GRBL's $13 parameter. This setting will switch GRBL's **reporting** to inches, io mm. It has no relevance for the Fusion PostProcessor

So let me summarize : 

1. GRBL accepps G20 / G21 command, and will then interpret your Gcode values as either inches or mm.
2. When drawing in Fusion, you choose units there, and this will result in Fusion sending a G20 or G21 during the post-process.
3. So if your units in Fusion are correct, it will send the correct G20 or G21 and distances in GRBL will be consistent with your design.

However, it is easy to import a drawing from anywhere, and not be aware of its units, then send this to  your CNC and all dimensions will be a factor 25.4 larger or smaller. Normally you always work in inches, or always in mm...
So I've built a little check into the PostProcessor to check for 'foreign, unexpected' units. (basically the 'other' unit from the one you are used to work with). It is done in the .cps file on line 42 / 43

```
var GRBLunits = MM;
// var GRBLunits = IN;

```
Further down the proces, when Fusion sends a G20 or G21, this is cross-checked with this setting : 

```
if (unit != GRBLunits)
	{
	if (GRBLunits == MM)
		{
		alert("Error", "GRBL configured to mm - CAD file sends Inches! - Change units in CAD/CAM software to mm");
		error("Fatal Error : units mismatch between CADfile and GRBL setting");
		}
	else
		{
		alert("Error", "GRBL configured to inches - CAD file sends mm! - Change units in CAD/CAM software to inches");
		error("Fatal Error : units mismatch between CADfile and GRBL setting");
		}
	}
```

So it is clear why you are getting this warning dialogbox : you are working in inches, but the PostProcessor expects mm... The warning messages are not 100% correct as it is not due to GRBLs settings, but rather a setting in the PostProcessor itself.

How to solve it ? Change the .cps file on lines 42 / 43 : outcomment line 42 and remove the // from line 43. Now your PostProcessor will expect inches

```
// var GRBLunits = MM;
var GRBLunits = IN;

```

I will improve the PostProcessor now that I am better understanding the $13 setting. It will take some time however, as I have a long stack of on-going projects and I need to fnish some of them first before embarking something new :-)

Strooom, 2017-12-16:
One more thing : you will need to adjust your machineHomeX, Y and Z parameters to values in inches.
Currently I see -10 in your settings. -10 **mm** is ok as a ''machine home' position, but it's certainly not -10 **inches**.
I guess you need something like -0.2.
See the other discussions on this Git repository about machineHome, G53 and its settings...