Strooom/GRBL-Post-Processor #20: Wiki pages can't be edited?

justinclift, 2019-06-01:
Just went to update a busted link on the wiki home page:

* https://github.com/Strooom/GRBL-Post-Processor/wiki

However wiki pages don't seem to be editable for this project?

Any chance of enabling editing, so it's easy for people to fix/update things? :smile:

Reference info, in case that's useful:

* Broken link: https://camforum.autodesk.com/index.php?topic=7571
  * It has a (very old) expired TLS certificate, so is unlikely to ever be fixed.
* Updated link: https://forums.autodesk.com/t5/hsm-post-processor-forum/g2-move-has-too-much-error/td-p/6087663
  * The above link just bounces here now anyway, so might as well use it.

Strooom, 2019-06-01:
Thanks Justin,

I don’t get any broken links, but I’ll remove the link to the forum as this
issue is now understood and solved.

Thanks for using and helping on the PostProcessor.

I’m currently working on an alternative to GRBL. This may be useful for you
too.
I’ll publish a first commit in a few weeks. I’m currently testing the
homing function.

Thanks,

P

justinclift, 2019-06-01:
No worries.  Stumbled on this via the Shapeoko Community forums:

&nbsp; &nbsp; https://community.carbide3d.com/t/new-electronics-again/14529/14

People there are saying good things, so it might be useful to create an account there and intro yourself when you have something for people to try? :smile:

closed 2019-06-04 by justinclift